Jashcraf/poke #28: Desired Features

Jashcraf, 2022-08-29:
Please leave a description of what you want to see in Poke in the future

Jashcraf, 2023-03-03:
# Features for LOFT Discussion

## Raytracing
- Support for Non-sequential ray traces in ZOS
- Support for Mixed-mode ray traces in ZOS

## Physics
- Accelerated GBD Algorithm
- Truncated Gaussian Beamlet Decomposition
- User-defined fresnel coefficients as a `coating` for surfaces

## Plotting
- "Publication-grade" default plotstyles for Jones Pupils, GBD Fields

## Exporting
- A clear exporting method for returning POPPY and HCIPy wavefronts from field data generated by poke
- Data structure to export and import rayfronts that are <not> pickles
- Data structure to write Jones Matrices and GBD fields to fits files

## Tests
- Going through the code and completing tests for raytrace, poke_core, thinfilms, writing, etc.
##

trentjbrendel, 2023-03-03:
## Reverse Ray Tracing / Stray Light Things
* Ability to trace non-sequential rays from the image plane into object space
* Ability to obtain a plot of the point source transmittance along an axis
* Scatter models (???)