Jashcraf/poke #29: 6.4 Compare GBD against other design codes

Jashcraf, 2022-09-05:
## Simulation Parameters

|     **Specification**    | **Value** |
|:------------------------:|:---------:|
|        wavelength        |   1.65um  |
|           npix           |    256    |
|        pixelscale        | 3.90625um |
| beamlets across aperture |    101    |

## Optical System is the HST
[image]

## Experiment

I'd like to use various optical modeling codes with coherent capabilities in order to determine how well the PSF is reconstructed. Ideally this simulation is done both with and without the Hubble aperture obscuration.

- [ ] Poke/GBD
- [ ] POPPY Fresnel w/ Zemax wavefront
- [ ] CODE V Beam Synthesis Propagation
- [ ] FRED Coherent Analysis

trentjbrendel, 2022-09-05:
@Jashcraf, could you clarify the details of the beam you want me to inject into the optical system? Each of these codes handles this slightly differently and I want to make sure I'm simulating the right thing. Thanks!

Jashcraf, 2022-09-05:
Sure - I'm looking for a collimated beam from infinity on-axis, so a truncated plane wave with 0 field value w.r.t. the optical axis.

If I recall, they both accept some parameters to constrain the beamlet calculation. I think FRED gives the user more control than CODE V does, so if "overlap factor" or something similar is available, set it to 1.7.

Also (and I'll add to the list above), that these should be **Irradiance** calculations. Intensity is also fine, just need the units to be Power instead of field. The phase doesn't matter so much here.

@trentjbrendel does that clarify it a bit better?

Jashcraf, 2022-09-05:
Oh also @trentjbrendel we should synch up on the right aperture to use, so just simulate the unobscured circular entrance pupil for now.

Jashcraf, 2022-09-08:
Top is GBD v.s. Analytical Airy, bottom is Zemax Huygens v.s. Analytical Airy
[image]

Jashcraf, 2022-09-08:
@trentjbrendel thoughts on this figure? Need to add titles but anything else?

trentjbrendel, 2022-09-10:
> 

Those figures look great! Is this the format you want me to use for reporting my results?

Jashcraf, 2022-09-11:
Nah I just need the raw data from you, frequently the centering of different diffraction methods is not the same so sometimes I need to shift things around. Also here's some titles.
[image]

closed 2023-08-19 by Jashcraf