Jashcraf/poke #49: Finalize Jones Pupil to FITS file conversion

Jashcraf, 2023-01-30:
poke.writing.JonesToFITS() needs some updating with keywords that are meaningful.

Jashcraf, 2023-01-30:
I think just some simple headers for wavelength + pixelscale would be helpful

Jashcraf, 2023-10-22:
@kenjim21 here's another fairly self-contained project that requires actually using Poke and stuff. I might need to generate some data for you.

Poke creates numpy arrays containing "Jones Pupil" data, which kind of tells you how the polarization state evolves through a given optical system (like a telescope!). It'd be convenient to transfer these numpy arrays with a file format that telescopey people use, so having a dedicated function in `poke.writing` that does this would be ideal.

Enter the FITS file format: https://fits.gsfc.nasa.gov/fits_primer.html (primer)

This is sort of used as a catch-all file transfer format in astronomy, because we work with a lot of data that takes the shape of images. The big takeaways are that the aforementioned numpy arrays are the "images" that I want to save. And I want them to include a header full of information about these data.

I've started working on this but it's barely functional: https://github.com/Jashcraf/poke/blob/a15947c959ded991f1d152927fc24c1edfc5ba42/poke/writing.py#L97

**Would you be willing to complete this function?** If so, give the FITS file primer a read and we can start talking about how to do a good translation with Poke.

Jashcraf, 2023-10-31:
@kenjim21 you said you had some Q's on this part, would you like to ask them here or set up a meeting to discuss the more physics-y side of Poke?

Jashcraf, 2023-10-31:
Totes lets do it! I'll send ya a message to set some stuff up. I'm also including below a list of notes for this feature that I thought of

Desired Header keywords
- pixel scale in pupil
- wavelength
- field of view
- raytrace filename (optional)
- polynomial decomposition fit error

Methods
- if regularly spaced in the entrance pupil, just pass those coordinates
- if not, do a zernike decomposition and report the fit error

Jashcraf, 2023-11-03:
Adding to the headers a name for each dimension of the numpy array

Jashcraf, 2023-11-03:
Have @kenjim21 write a demo and test file showing these things off

Jashcraf, 2023-11-08:
@kenjim21 Just added what you'll need to get started in the most recent commit to branch [49-finalize-jones-pupil-to-fits-file-conversion](https://github.com/Jashcraf/poke/tree/49-finalize-jones-pupil-to-fits-file-conversion)

It comes with a more fleshed out `poke.writing.jones_to_fits` function, and more importantly the beginnings of a Demo that I was hoping you could flesh out. You can find it in `poke/docs/notebooks/jones_to_fits.ipynb`

It basically shows you how to load a sample Rayfront and grab its Jones pupil. This should let you play around with some data.

kenjim21, 2023-11-20:
@Jashcraf Just finished updating the jones_to_fits function and the notebook for base demonstration. Let me know what you think of this as a start and we can build up from there!

kenjim21, 2023-12-21:
@Jashcraf Just made some minor style changes for the demonstration notebook. Let me know what you think!

Jashcraf, 2023-12-22:
Looks good I think we are basically ready to make a PR for this feature. There's just a couple things I would like edited:
```
1. realimag: defaults to true (meaning imaginary values will be saved as opposed to saving the absolute values)
```
Should instead be something like
```
1. realimag: defaults to true (meaning complex values are saved as real and imaginary components, if false, saves as absolute value and phase)
```

And for anything in text that is expressing something that you write in code, could you format it so that it looks like code? Example:
- Primary Header (hdul[0]):
- Primary Header (`hdul[0]`)

Other places where this change should be made include `import astropy.io`, the FITS headers, and the `with fits.open('sample_jones.fits') as hdul:`.

After that make a PR and I think we are set :) Thanks for doing this!

closed 2023-12-22 by Jashcraf