Jashcraf/poke #51: Rayfront.opd analysis functions

Jashcraf, 2023-03-07:
We are storing the optical path difference of the rays in both CODE V and Zemax but need a way to actually make the data meaningful. As of now it's a raveled array of data, but diffraction calculations are done on square grids!

So, we need
- method of decomposing the WFE into polynomials (Zernikes are easy)
- method of differentiating ray OPD with OPD data
- relevant analysis of OPD

Hyukmo was interested in this utility so we can work with him on that.

Jashcraf, 2023-03-07:
@hyukmo I do think priority #1 is coming up with some code to do zernike decomposition in poke

do you think you can write some code in python to generate a zernike polynomial given some index?

hyukmo, 2023-03-07:
@Jashcraf I can try, captain. But it would take some time....

Jashcraf, 2023-03-07:
No worries just need the contribution before SPIE haha

we can either write our own, or make poke dependent on another package. If possible i think it'd be nice to do our own for low order polynomials (say Z1-36 or whatever). I can also help out with it and we can use this issue to workshop the code.

trentjbrendel, 2023-03-15:
Jaren, how much do we need to generate code from scratch? Would it be okay if I prompted ChatGPT for some boilerplate code?

Jashcraf, 2023-03-15:
le mao
I have a preference against it so that contributors can use this as a learning experience, also I have no clue how to cite ChatGPT or it's products. 

but zernike decomposition is just a problem of generating the polynomials, [which wikipedia tells you how to do](https://en.wikipedia.org/wiki/Zernike_polynomials), and computing the dot product of them with your data.

Jashcraf, 2023-03-23:
Turns out it might be better to use `np.linalg.lstq` to solve for the coefficient vector!

Jashcraf, 2023-08-19:
Emory added the ability to do Zernike decomposition, so now we can do that with OPD data. Might be worth the assertion that OPD has consistent units in poke.raytrace.

closed 2023-10-31 by Jashcraf