Jashcraf/poke #54: Debugging Inconsistent field on translation

Jashcraf, 2023-04-05:
r_ray and k_ray look good to me, let's try the ray transfer matrices.

# A matrix
[image]

# B Matrix
[image]

# C matrix
[image]

# D matrix
[image]

Jashcraf, 2023-04-05:
top is new algo, bottom is old

overall the new one looks less regular, maybe an operation got broadcast over an unintended dimesnion

Jashcraf, 2023-04-05:
Delta calculation
[image]

These actually just look like the right values in perhaps different places. Specifically - the middle two columns are switched

Jashcraf, 2023-04-05:
# K rays before delta calculation are the same, but different order
[image]

Jashcraf, 2023-04-05:
Oh my god lData and mData were flipped in the function inputs

Jashcraf, 2023-04-05:
Delta is good (left gbd, right beamlets)
[image]

Jashcraf, 2023-04-05:
# After orthogonal transformation
k_rays on both == np.array([0,0,1]), which is good!
r_rays look a little different at first glance, can we use fewer beamlets without breaking?

[image]

Jashcraf, 2023-04-05:
Gonna look at the +X differential ray, r_rays should basically all be at x,y = 0 here.
They are different and transposed!
[image]

Jashcraf, 2023-04-05:
Ah I think I might have been looking at the wrong coordinate
here is the x,y,z position for each ray at a given pixel - which it is constant across
[image]
consistency!

Jashcraf, 2023-04-05:
plotting after the delta propagation, the beamlets array is entirely black?
[image]

Jashcraf, 2023-04-05:
Oh it's because I'm passing it weird data, the rayset index is now in the first when it should be in the zeroth. 
consistency!
[image]

Jashcraf, 2023-04-05:
Checking after orthogonal transformation. Might have some nan vals in the old method. Viridis is r_data and magma is k_data
[image]
[image]

Both look good (save the nans)

Jashcraf, 2023-04-05:
# Axx
[image]

# Cxx - looks wrong
[image]

# Bxx - looks wrong, values of near 50 are about what we'd expect given the HST focal length
[image]

# Dxx - also doesn't look spectacular
[image]

Jashcraf, 2023-04-05:
Ah I forgot to normalize the l vector in the orthogonal transfer matrix

# Axx - looking good!
[image]
# Cxx - looking good!
[image]
# Bxx - looking good!
[image]
# Dxx - tell me why im so excited
[image]

Jashcraf, 2023-04-05:
WE WIN
[image]

time to try the misaligned case

Jashcraf, 2023-04-05:
Breaks during misalignment and focus - I suspect that this has to do with where the ray positions are
[image]

Jashcraf, 2023-04-05:
[image]

Nah the ray positions are identical

Jashcraf, 2023-04-05:
Let's check the final ingredients for the matrix, starting with Amplitude
# |Amplitude|
[image]
Looks slightly different, don't know what this could be from (other than the complex beam parameter computation)

# Angle{Phase}
This is very different!
[image]

(The vertical lines are the new algorithm)

Jashcraf, 2023-04-05:
For the misaligned case the ray transfer matrices are different!
# Axx (left is beamlets right is gbd)
[image]

Jashcraf, 2023-04-05:
Works for the defocused case
[image]

Jashcraf, 2023-04-05:
The orthogonal transformation matrix is looking good, each image is a row in the matrix
[image]
[image]
[image]

Jashcraf, 2023-04-05:
[image]
The detector coords are wrong!

Jashcraf, 2023-04-05:
[image]
turns out you need to provide the offset to the rays, not detector

trentjbrendel, 2023-04-05:
This is fascinating. Thanks for sharing, Jaren!

closed 2023-04-06 by Jashcraf