Jashcraf/poke #60: Testing poke on the SPT-SLIM Radio Instrument

trentjbrendel, 2023-04-11:
# Using poke to test physical optics for radio wavelengths
Frequency: 150 GHz
Wavelength: 2.0 mm

```
nrays = 20
wavelength = 2.0e-3
pupil_radius = 5.0
max_fov = 0.04
OF = 1.5
wo = 2*pupil_radius*OF / (2*nrays)
```
```
dsize = 50e-3
npix = 256
```
[image]

Currently working on varying the pupil_radius, max_fov, nrays, dsize, and npix to explore the solution space in an attempt to match the results from poke with the Huygens PSF results from Zemax.

Will update when I have more results.

trentjbrendel, 2023-04-11:
```
import sys
sys.path.append('C:/Users/trent/OneDrive/Documents/GitHub/poke-restructure/poke')

# On YOGA:
# C:\Users\trent\Documents\GitHub\poke-restructure\poke

# On CyberpowerPC:
# C:\Users\trent\OneDrive\Documents\GitHub\poke-restructure\poke

from poke.poke_math import set_backend_to_numpy
set_backend_to_numpy()
from poke.poke_math import np
from poke.poke_core import Rayfront
import matplotlib.pyplot as plt
from astropy.io import fits
import sys
import pickle
import poke.plotting as plot

# pth = 'C:/Users/trent/Dropbox/WORK/RESEARCH/SPT-SLIM/SPT-SLIM_Forward_F2_ReDesign_v6_FIN.zmx'
pth = 'S:/Dropbox/WORK/RESEARCH/SPT-SLIM/SPT-SLIM_Forward_F2_ReDesign_v6_FIN.zmx'
nrays = 40
wavelength = 2.0e-3
pupil_radius = 5.0
max_fov = 0.04
OF = 1.5
wo = 2*pupil_radius*OF / (2*nrays)
print('waist size [m] = ',wo)

# set up surfaces
s5 = {
    'surf':5,
    'mode':'reflect',
    # 'coating': 0.04 + 1j*7
    #0.04 + 1j*7
}
# s11 = {
#     'surf':11,
#     'mode':'reflect',
#     'coating': 0.04 + 1j*7
# }
# s19 = {
#     'surf':19,
#     'mode':'reflect',
#     'coating': 0.04 + 1j*7
# }
# s22 = {
#     'surf':22,
#     'mode':'reflect',
#     'coating': 0.04 + 1j*7
# }
si = {
    'surf':47,
    'mode':'refract',
#     'coating':0.0
}
surflist = [s5,si]

# set up detector coordinates
dsize = 100e-3
npix = 512
x = np.linspace(-dsize/2,dsize/2,npix)
x,y = np.meshgrid(x,x)
dcoords = np.asarray([x.ravel(),y.ravel(),0*x.ravel()])

rf = Rayfront(nrays,wavelength,pupil_radius,max_fov,waist_pad=wo,normalized_pupil_radius=0.5)
rf.as_gaussianbeamlets(wo)
rf.trace_rayset(pth,surfaces=surflist)
# plot.RayOPD(rf)
# with open ('test_hst_rayfront_gauslets_30beams_1.65um_zmx.pickle','rb') as f:
#     rf = pickle.load(f)

field = rf.beamlet_decomposition_field(dcoords,memory_avail=40).reshape([npix,npix])

plt.figure()
plt.imshow(np.log10(np.abs(field)**2))
plt.colorbar()
plt.show()
```

[image]

Jashcraf, 2023-10-31:
This was an excellent experiment, many thanks @trentjbrendel ! Gonna close it for now since the SPT-SLIM time is over

closed 2023-10-31 by Jashcraf