Jashcraf/poke #89: fixed dependency versions hurt my soul

brandondube, 2023-08-22:
I created a conda environment for poke:
```
$ conda create -n poke python=3.11 msgpack-python jupyterlab numpy scipy matplotlib astropy
```

Then upon pip install, it clobbered a lot of those with pinned versions:

<details><summary><code>$ pip install -e .</code></summary><br><pre>
Obtaining file:///Users/bdube/src/poke
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... done
  Preparing editable metadata (pyproject.toml) ... done
Requirement already satisfied: numpy>=1.23.5 in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from poke==1.0) (1.25.2)
Requirement already satisfied: matplotlib>=3.6.2 in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from poke==1.0) (3.7.1)
Collecting astropy>=5.1.1 (from poke==1.0)
  Obtaining dependency information for astropy>=5.1.1 from https://files.pythonhosted.org/packages/5a/f7/c7030f9d1190bcc49564432a4cb536494aecd7d3fcb654cc5d604c802607/astropy-5.3.2-cp311-cp311-macosx_11_0_arm64.whl.metadata
  Downloading astropy-5.3.2-cp311-cp311-macosx_11_0_arm64.whl.metadata (9.4 kB)
Collecting zosapi==0.2 (from poke==1.0)
  Downloading zosapi-0.2.tar.gz (3.4 kB)
  Preparing metadata (setup.py) ... done
Collecting numexpr>=2.8.4 (from poke==1.0)
  Obtaining dependency information for numexpr>=2.8.4 from https://files.pythonhosted.org/packages/00/7b/e9cbb3a7fd7baa3fadfa1e710fdd6be939d106d25058b43ea49da1ec6229/numexpr-2.8.5-cp311-cp311-macosx_11_0_arm64.whl.metadata
  Downloading numexpr-2.8.5-cp311-cp311-macosx_11_0_arm64.whl.metadata (8.0 kB)
Collecting msgpack>=1.0.5 (from poke==1.0)
  Downloading msgpack-1.0.5-cp311-cp311-macosx_11_0_arm64.whl (69 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 69.0/69.0 kB 4.9 MB/s eta 0:00:00
Collecting msgpack-numpy>=0.4.8 (from poke==1.0)
  Downloading msgpack_numpy-0.4.8-py2.py3-none-any.whl (6.9 kB)
Collecting numpydoc==1.5.0 (from poke==1.0)
  Downloading numpydoc-1.5.0-py3-none-any.whl (52 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 52.4/52.4 kB 4.9 MB/s eta 0:00:00
Collecting nbsphinx==0.8.11 (from poke==1.0)
  Downloading nbsphinx-0.8.11-py3-none-any.whl (26 kB)
Collecting docutils (from nbsphinx==0.8.11->poke==1.0)
  Obtaining dependency information for docutils from https://files.pythonhosted.org/packages/26/87/f238c0670b94533ac0353a4e2a1a771a0cc73277b88bff23d3ae35a256c1/docutils-0.20.1-py3-none-any.whl.metadata
  Downloading docutils-0.20.1-py3-none-any.whl.metadata (2.8 kB)
Requirement already satisfied: jinja2 in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from nbsphinx==0.8.11->poke==1.0) (3.1.2)
Requirement already satisfied: nbconvert!=5.4 in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from nbsphinx==0.8.11->poke==1.0) (6.5.4)
Requirement already satisfied: traitlets>=5 in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from nbsphinx==0.8.11->poke==1.0) (5.7.1)
Requirement already satisfied: nbformat in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from nbsphinx==0.8.11->poke==1.0) (5.7.0)
Collecting sphinx>=1.8 (from nbsphinx==0.8.11->poke==1.0)
  Obtaining dependency information for sphinx>=1.8 from https://files.pythonhosted.org/packages/40/c0/e62ce9d243bfa2d9f290d7c730c7df3d9e979f0016e38c767f33ffcc2185/sphinx-7.2.2-py3-none-any.whl.metadata
  Downloading sphinx-7.2.2-py3-none-any.whl.metadata (5.9 kB)
Collecting pythonnet (from zosapi==0.2->poke==1.0)
  Downloading pythonnet-3.0.1-py3-none-any.whl (284 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 284.5/284.5 kB 12.9 MB/s eta 0:00:00
Requirement already satisfied: pyerfa>=2.0 in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from astropy>=5.1.1->poke==1.0) (2.0.0)
Requirement already satisfied: PyYAML>=3.13 in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from astropy>=5.1.1->poke==1.0) (6.0)
Requirement already satisfied: packaging>=19.0 in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from astropy>=5.1.1->poke==1.0) (23.0)
Requirement already satisfied: contourpy>=1.0.1 in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from matplotlib>=3.6.2->poke==1.0) (1.0.5)
Requirement already satisfied: cycler>=0.10 in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from matplotlib>=3.6.2->poke==1.0) (0.11.0)
Requirement already satisfied: fonttools>=4.22.0 in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from matplotlib>=3.6.2->poke==1.0) (4.25.0)
Requirement already satisfied: kiwisolver>=1.0.1 in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from matplotlib>=3.6.2->poke==1.0) (1.4.4)
Requirement already satisfied: pillow>=6.2.0 in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from matplotlib>=3.6.2->poke==1.0) (9.4.0)
Requirement already satisfied: pyparsing>=2.3.1 in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from matplotlib>=3.6.2->poke==1.0) (3.0.9)
Requirement already satisfied: python-dateutil>=2.7 in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from matplotlib>=3.6.2->poke==1.0) (2.8.2)
Requirement already satisfied: MarkupSafe>=2.0 in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from jinja2->nbsphinx==0.8.11->poke==1.0) (2.1.1)
Requirement already satisfied: lxml in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from nbconvert!=5.4->nbsphinx==0.8.11->poke==1.0) (4.9.2)
Requirement already satisfied: beautifulsoup4 in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from nbconvert!=5.4->nbsphinx==0.8.11->poke==1.0) (4.12.2)
Requirement already satisfied: bleach in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from nbconvert!=5.4->nbsphinx==0.8.11->poke==1.0) (4.1.0)
Requirement already satisfied: defusedxml in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from nbconvert!=5.4->nbsphinx==0.8.11->poke==1.0) (0.7.1)
Requirement already satisfied: entrypoints>=0.2.2 in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from nbconvert!=5.4->nbsphinx==0.8.11->poke==1.0) (0.4)
Requirement already satisfied: jupyter-core>=4.7 in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from nbconvert!=5.4->nbsphinx==0.8.11->poke==1.0) (5.3.0)
Requirement already satisfied: jupyterlab-pygments in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from nbconvert!=5.4->nbsphinx==0.8.11->poke==1.0) (0.1.2)
Requirement already satisfied: mistune<2,>=0.8.1 in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from nbconvert!=5.4->nbsphinx==0.8.11->poke==1.0) (0.8.4)
Requirement already satisfied: nbclient>=0.5.0 in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from nbconvert!=5.4->nbsphinx==0.8.11->poke==1.0) (0.5.13)
Requirement already satisfied: pandocfilters>=1.4.1 in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from nbconvert!=5.4->nbsphinx==0.8.11->poke==1.0) (1.5.0)
Requirement already satisfied: pygments>=2.4.1 in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from nbconvert!=5.4->nbsphinx==0.8.11->poke==1.0) (2.15.1)
Requirement already satisfied: tinycss2 in /Users/bdube/miniconda3/envs/poke/lib/python3.11/site-packages (from nbconvert!=5.4->nbsphinx==0.8.11->poke==1.0) (1.2.1)
[... 65 lines cut ...]
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 83.5/83.5 kB 9.5 MB/s eta 0:00:00
Downloading sphinxcontrib_qthelp-1.0.6-py3-none-any.whl (89 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 89.4/89.4 kB 7.6 MB/s eta 0:00:00
Building wheels for collected packages: poke, zosapi
  Building editable for poke (pyproject.toml) ... done
  Created wheel for poke: filename=poke-1.0-0.editable-py3-none-any.whl size=3694 sha256=ca651d35d5a422a5418931019f48e46755a2ce4fa2eb3e9f26dae22097915a69
  Stored in directory: /private/var/folders/hw/82bzy1hn5zxdlfrksnvdmqf40000gq/T/pip-ephem-wheel-cache-f5wxa_ee/wheels/68/a6/2e/f1c1633df3038aa39d97de13ceb98190ef152a9c92747e56cb
  Building wheel for zosapi (setup.py) ... done
  Created wheel for zosapi: filename=zosapi-0.2-py3-none-any.whl size=3632 sha256=050ed499f6b325737f0f69d65546bce99f807138ee38b8ef1625e4622a7a438b
  Stored in directory: /Users/bdube/Library/Caches/pip/wheels/08/d2/50/ab3479d2ab1a3aad59b06ce65908c375c13f0205e46df2ff98
Successfully built poke zosapi
[... cut ...]
Successfully installed alabaster-0.7.13 astropy-5.3.2 clr-loader-0.2.6 docutils-0.20.1 imagesize-1.4.1 msgpack-1.0.5 msgpack-numpy-0.4.8 nbsphinx-0.8.11 numexpr-2.8.5 numpydoc-1.5.0 poke-1.0 pythonnet-3.0.1 snowballstemmer-2.2.0 sphinx-7.2.2 sphinxcontrib-applehelp-1.0.7 sphinxcontrib-devhelp-1.0.5 sphinxcontrib-htmlhelp-2.0.4 sphinxcontrib-jsmath-1.0.1 sphinxcontrib-qthelp-1.0.6 sphinxcontrib-serializinghtml-1.1.9 zosapi-0.2

</pre></details>

It would be lovely if poke specified at worst lower bounds on versions, so that installing poke doesn't potentially change/break existing environments

Jashcraf, 2023-08-22:
Roger that and can do! Would upper and lower bounds be better?

brandondube, 2023-08-22:
upper bounds accelerate the heat death of the universe*, lower only please

\* mean that eventually your code can't be used with anything else unless you religiously update the dependency versions, which you probably won't experience any breaking changes from anyway (in ~10y, not pinning numpy and scipy versions has never hurt me)

Jashcraf, 2023-09-25:
Mostly fixed a while ago, finished up everything in 884a2dd049468bf2d84a37381f17bb07d681ac55

closed 2023-09-25 by Jashcraf