Jashcraf/poke #109: msgpack_numpy not found

jkentwallace, 2024-08-31:
Hi Jaren, 
I'm getting the following error while running POKE
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-5-2dcd5e2cf0e1> in <module>
      1 import numpy as np
      2 import matplotlib.pyplot as plt
----> 3 from poke.writing import read_serial_to_rayfront
      4 from poke.poke_core import Rayfront

~/Documents/Compact Coronagraph - Low Order Control/00 SAT Compact Coronagraph and Low Order Control/04 Performance Modeling/01_Polarization_Phase_Delay/01POKE/poke/poke/writing.py in <module>
      1 import numpy as np
      2 import msgpack
----> 3 import msgpack_numpy as m
      4 from poke.poke_core import Rayfront
      5 from poke.interfaces import regularly_space_jones

ModuleNotFoundError: No module named 'msgpack_numpy'

Jashcraf, 2024-08-31:
> *External Email*

Ah might need to add msgpack_numpy to the requirements.txt

Try ‘pip install msgpack_numpy’

jkentwallace, 2024-08-31:
Hey! That was fast! Here's what I did: 
===
(base) kwallace@MT-210368 poke % pip install msgpack-numpy
Collecting msgpack-numpy
  Downloading msgpack_numpy-0.4.8-py2.py3-none-any.whl (6.9 kB)
Requirement already satisfied: numpy>=1.9.0 in /Users/kwallace/opt/anaconda3/lib/python3.7/site-packages (from msgpack-numpy) (1.18.1)
Requirement already satisfied: msgpack>=0.5.2 in /Users/kwallace/opt/anaconda3/lib/python3.7/site-packages (from msgpack-numpy) (0.6.1)
Installing collected packages: msgpack-numpy
Successfully installed msgpack-numpy-0.4.8
===
Now, I'm on to the next error: 

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-6-4143196c87a6> in <module>
      1 # Load a rayfront
      2 pth_to_rf = '/Users/kwallace/Documents/Compact Coronagraph - Low Order Control/00 SAT Compact Coronagraph and Low Order Control/04 Performance Modeling/01_Polarization_Phase_Delay/01POKE/poke/tests/ELT_rayfront_aspolarized_64rays_0.6um.msgpack'
----> 3 rf = read_serial_to_rayfront(pth_to_rf)
      4 display(rf.surfaces)
      5 

~/Documents/Compact Coronagraph - Low Order Control/00 SAT Compact Coronagraph and Low Order Control/04 Performance Modeling/01_Polarization_Phase_Delay/01POKE/poke/poke/writing.py in read_serial_to_rayfront(filename)
     97         serdata = infile.read()
     98 
---> 99     rayfront = deserialize(serdata)
    100 
    101     return rayfront

~/Documents/Compact Coronagraph - Low Order Control/00 SAT Compact Coronagraph and Low Order Control/04 Performance Modeling/01_Polarization_Phase_Delay/01POKE/poke/poke/writing.py in deserialize(buf)
     57     Tname, varzzz = msgpack.unpackb(buf, use_list=True)
     58     for k, v in varzzz.items():
---> 59         setattr(e, k, v)
     60 
     61     e.__class__ = globals()[Tname]

TypeError: attribute name must be string, not 'bytes'

Jashcraf, 2024-08-31:
> Hey! That was fast! Here's what I did: (base) ***@***.*** poke %

Oh I havent seen this before, must be a version thing…

Im about to get on a plane to LA, but I can look into this tomorrow. In the
meantime, try going back a version of ‘msgpack’ and ‘msgpack-numpy’

you can do this by ‘pip install msgpack==0.0.0’ to install version 0.0.0 of
the mspack package. (Don’t actually do version 0.0.0 though haha)

jkentwallace, 2024-08-31:
Sounds good! I'll give it a shot. 
Traveling mercies to you! 
You can anticipate this kinda of pestering for a while...thanks for your patience...