Jashcraf/poke #113: small error + feature request

kira-hart, 2025-01-02:
in line 288 in prt_matrix I believe 
`fss, fps = surfdict["coating"](aoi)`
should be
`fss, fpp = surfdict["coating"](aoi)`

https://github.com/Jashcraf/poke/blob/fcd2109841908b84dd1a754823af30a683ecb72e/poke/polarization.py#L288C14-L288C18

Also, a relatively simple feature add that would be fantastic is the ability to supply a user defined jones pupil (as a callable function) as a coating.

Jashcraf, 2025-01-08:
Thanks for pointing this out! If you'd like to make the PR I'd be happy to merge so you get credit for the contribution. Otherwise I can make the fix.

Jashcraf, 2025-01-29:
@kira-hart just getting back on this - were you interested in making the PR to fix this issue? If not, I can try get it done by end of week