Jashcraf/poke #115: Polarization ray tracing - Zemax sequential mode

kalosu, 2025-01-06:
Hello there Poke community,

I have managed to install Poke on my computer and have also managed to run an initial ray tracing analyzis following the given examples. To be more precise, I have taken the following example as reference:

https://github.com/Jashcraf/poke/blob/main/examples/gen_jonespupil.py

From this I have a few questions:

-First, I was wondering if there is any effect from the radius and ffov parameters given in the example. In my case, no matter what value I use, I always get the same coordinates in the pupil. Also, does this matter? At the end, I see that the pupil coordinates are always normalized and this makes me think that the radius and ffov parameters are not important, i.e, the system parameters defined from within Zemax determines the pupil coordinates?

-In my case, at the moment, I am just modeling the case of light being refracted by a Weierstrass lens and I try to determine the Jones pupil after this surface. You can see an example of this system here (I have more surfaces in my system but for now I am just limiting myself to the first Weierstrass lens):

[image]

Once I use ```rf.compute_jones_pupil(aloc=np.array([0.0,0.0,-1.0]),collimated_object=False)```, I get the following plots for the Jones pupil:

[image]

To me it looks a bit weird that the magnitude of the Jones pupil entries is not symmetric, specifically I am referring to the upper part of the magnitude plots. How could I validate these results? What could be happening here?

Finally, this is the code that I am using:
```
nrays = 64
wl = 640e-9
radius = 1
ffov = 70.0 ##In degrees

n1 = 1.0
n2 = n_IPVisio_MS(0.640).real ##Some finite float value 

rf = Rayfront(nrays,wl,radius,ffov)

s1 = {
    'surf':1,
    # 'coating':[n2,n1],
    'coating':[n2,n1],
    'mode':'transmit'
}

rf.as_polarized([s1])

rf.trace_rayset(path)

###We use the polarization ray tracing part
rf.compute_jones_pupil(aloc=np.array([0.0,0.0,-1.0]),collimated_object=False)

pl.jones_pupil(rf)
```

Thanks a lot for the feedback and comments!

kalosu, 2025-01-06:
Hi there again,

I have found that if I change the value of my incident object cone angle to a lower value, then the asymmetries are gone. 
In my optical system, I set the aperture type as object cone angle with an original aperture value of 70 degrees (that corresponds to the plot I included in my first post).

If I decrease the object cone angle to a value of 40 degrees, then the asymmetries are not present anymore.

[image]

What could be happening here?  Any thoughts on what could I check for to find the issue?

kalosu, 2025-01-07:
Hello again,

I have been digging a bit more in order to find the cause for the issue I observed. While going through the function ```trace_through_zos``` I observed somethings that are not fully clear to me.  In my case, I have only access to the latest Ansys Zemax version and after using ```BatchRayTrace``` in combination with ```rays = reader.InitializeOutput(nrays)``` I was trying to inspect the content stored within ```rays```.

For this I continue to use my single Weierstrass lens geometry example. When I plot the direction cosines incident towards my single surface, I notice that these are expressed in terms of the local coordinates basis. If I compute the angle made between the xy components of k with respect to the z component, then the angles I obtain seem to agree with the angles made with respect to the local z coordinate to each ray:

[image]

I verified that these vectors are expressed in terms of local coordinates by replacing my Weierstrass lens with just a planar aperture. In the case of the planar aperture (with constant surface normal), the obtained angles between the xy and z components of k yield the correct angle in global coordinates (in this case, this is true since N equal to (0,0,1) for each point in the aperture).

I noticed that within the ```trace_through_zos``` function there is a part in where these local vectors are supposed to be re expressed in terms of the global coordinate system by means of the ```Rmat``` matrix. However in my case, the ```Rmat``` is equal to just a single global identity matrix. 

From my understanding, it is needed that the k vectors are expressed in terms of the global coordinate system.  (For the purpose of computing the transmitted k vectors and for the purpose of assembling the global rotation matrices associated to each P polarization ray tracing matrix). Would this be correct? 

Thanks a lot for your comments!

kalosu, 2025-01-07:
Hi there again,

I was still checking around with the issue that I posted first and somehow it seems that it is more of a visualization issue. 

If I replace the scatter plots from within the ```pl.jones_pupil(rf)``` function by ```tricontourf``` plots, then the asymmetries seem to dissapear. 

[image]

I am still not sure if these results are correct. Specifically, because of the way $k_{out}$ is constructed. The output ray directions are just the direction cosines returned by Zemax. However, these are expressed in the local coordinate basis for each ray, which as far as I understand it, corresponds to the coordinate system in where the local z is aligned with the surface normal at the given intersection point. Should $k_{out}$ not be expressed but in terms of the global coordinate system? Specially if I want to obtain $k_{in}$ from $k_{out}$, this would require that $k_{out}$ is expressed in the same basis as the $N$ vectors right? Any thoughts or comments?

Nevertheless, I then tried to use the computed $P$ matrices to calculate the output electric field from an input electric field distribution. In this case, I defined the local polarization state as just being linear along x and then used the double pole coordinate based transformation to obtain the electric field components associated to $k_{in}$ in terms of the global coordinates (this means, I obtain Ex, Ey and Ez).

After taking the product between $P$ and $E_{in}$, I got $E_{out}$ as: 
[image]

Somehow it seems that it is going in the right direction. Now in this case, both e_in and e_out are purely real. Do these components represent the magnitude of the $E_x$, $E_y$ and $E_z$ components or something else? How could I include the propagation phase in this case as well as the phase changes due to the polarization effects? At the end I want to have complex amplitude values that include the accumulated phase from the propagation across my system, so that I can use these field components to calculate a diffraction integral. 

Any thoughts about this? Thanks a lot for the feedback!!

Jashcraf, 2025-01-08:
Hello! Apologies for the delays - I've been traveling

It's very possible that it's an issue with `plt.scatter()`, which is what `plotting.jones_pupil()` uses. The discrete colorbar you used in your most recent example may remedy that a little. Some other things to try:
- make your own scatter plot, and decrease the dot size, i.e. `plt.scatter(x, y, c=data, s=size)`
- try tracing more rays: At high angles of incidence like in this example, small changes can result in visible asymmetries - which go away with higher sampling