Jashcraf/poke #116: Setting an incident field distribution

kalosu, 2025-01-13:
Hello there friends from Poke,

I was wondering if you have any suggestion on how should I set the incident electric field components for the case in where my polarization state is not initially described by Jones vectors.

I have the following optical system:

[image]

In this case, I have a point source embeded in some solid immersion lens. I have a lens to collect the light and another spherical surface to refocus the field.

For the case in where I want to model a point emitter that generates a linearly polarized field I can easly set this by using a jones vector in combination with the local coordinate axis vectors obtained from the double pole approach. 

In this case, from the 2D jones vectors, I can compute the 3D vectors that represent my linearly polarized field distribution with components Ex, Ey and Ez.

After running the ray-tracing, I use the polarization ray-tracing routines to obtain the total polarization matrix of the system.

For this specific case, for the linearly polarized wave, the results that I obtain by doing $P_{total} \cdot E_{in}$ agree with the results obtained from Zemax while doing a polarization ray-tracing in the nonsequential mode for a source with the same polarization.

So far, everything good. However, in my "real" case, I have a general electric field distribution obtained from a FDTD simulation. Here, similarly, I also have $E_x$, $E_y$ and $E_z$. However these field components are obtained from a near-to-far-field transformation.

Here the "assumption" that I use is that I am evaluating the field at the far-field, meaning that I extract the field components over the surface of a sphere at a sufficiently large distance from the origin. In this regime, the field has the nice property that it's shape does not change and just the amplitude of the field scales as a function of the distance from the origin.

Based on this assumption, I also took these field components as the input to the polarization ray-tracing for a point source (Essentially what I assume is that for each ray traced through the system, the field components associated to these rays is just the far-field amplitude of each planewave which is described by $E_x$, $E_y$ and $E_z$). 

However, in this case, I am not getting the same results as in the first example (globally linearly polarized).

I was wondering if you had any idea on what could I do here? The $P_{total}$ matrices account for amplitude changes due the refraction at the interfaces right? Any suggestion on how could I set the incident field distribution in this case?

Thanks a lot for the comments!!

kalosu, 2025-01-13:
Hi there again, 

Sorry I forgot one more thing. Does the $P_{total}$ include any effect on the amplitude change due to the propagation in between interfaces? When I take the dot product of $P_{total}$ with $E$, this only includes Fresnel effects right? What about the changes in field amplitude due to the propagation in between the interfaces? How could I set this change in amplitude as well?

At the final surface in where I need to evaluate the field distribution, I obtain the complex amplitude distributions from the field vector $E_{out}$ times the exponential that accounts for the free space propagation through the system via the optical path length phase. However, this does not account for the amplitude change due to the free-space propagation. How could I set this thing as well?

Again, thanks a lot for the comments!

Jashcraf, 2025-01-29:
> For this specific case, for the linearly polarized wave, the results that I obtain by doing Ptotal * Ein agree with the results obtained from Zemax while doing a polarization ray-tracing in the nonsequential mode for a source with the same polarization.

Woo! Glad to hear it is working thus far.

> However, in this case, I am not getting the same results as in the first example (globally linearly polarized). ... I was wondering if you had any idea on what could I do here? 

My first thought is that I'm not too surprised that a simulation with global linear polarization and some FDTD simulation would have dissimilar results. You are right on using Ptotal instead of the jones pupil here. I'd be curious to see just how different the data looks. Feel free to post here, or if you are worried about sharing data publicly, you can email me at jarenashcraft@ucsb.edu and I can give you some thoughts.

> The Ptotal matrices account for amplitude changes due the refraction at the interfaces right? Any suggestion on how could I set the incident field distribution in this case?

This is correct, they account for phase and amplitude changes due to the refraction at the interfaces. From what you describe, it sounds like you are doing the correct thing by dotting the E vector with the full PRT matrix, and not the Jones pupil. 

> What about the changes in field amplitude due to the propagation in between the interfaces? How could I set this change in amplitude as well?

Poke doesn't account for propagation amplitudes unfortunately. To do so, you would need to use a separate physical optics propagation software. [POPPY](https://poppy-optics.readthedocs.io/en/latest/) and [prysm](https://prysm.readthedocs.io/en/stable/) are two that close collaborators of mine develop for, if that is something you are interested in. However, it doesn't tend to play nice with terribly high NA systems. We can talk about the precise implementation if you want to try this method.