Jashcraf/poke #117: Field amplitude from ray amplitude information

kalosu, 2025-01-22:
Hi there Poke community, 

I was wondering if you had any suggestion on how could I obtain field amplitude information from the ray information that I generate with Poke. 

With Poke I can do the polarisation ray tracing to see how the polarisation state of my ray set changes through my optical systems. 

However, this does not directly translate in field amplitude changes due to the “free space propagation” (I know, to be fully consistent and correct this would actually require using using a diffraction integral based approach but for now I just want to evaluate the geometrical aspects only) and neither due to the change in ray density as my field propagates through my system 

As an example: let’s say I have a beam shaping surface which takes a Gaussian beam and generates a uniform intensity or irradiance disk at some far field plane.

If I  generate a ray set that represents my input Gaussian beam with a linear polarisation, I can use Poke to obtain the field after my surface, i.e, by including the possible changes in polarisation state (all of this is ray based information). I then use these ray amplitudes and propagate them geometrically to the desired target plane. In this case the ray amplitudes still follow the Gaussian distribution ( since after the transmission, probably the amplitude does not change that much, the shaping characteristic is introduced by the phase change by the refraction in the surface or in this case by adjusting the ray density at the target plane). However the field amplitude should be uniform at the target plane right? ( since the intensity or irradiance is the square of the field amplitude) 

How could I connect these two pieces of information? 
Or better said, how could I generate field amplitude distributions from the ray amplitude information? Do you have any suggestion in mind? (Again, I want to do this purely geometrically for the moment, while neglecting diffraction and coherence effects) 

Thanks a lot for your comments!!

Jashcraf, 2025-01-27:
Pardon me was traveling and then on vacation - I'll pick up this issue shortly

Jashcraf, 2025-01-29:
My response to your other issue #116 included a comment at the end to introduce free-space propagation amplitude effects if you are interested in those. Though incorporating them in your simulation is non-trivial given the high-NA object, so it would warrant some further discussion on that.

> As an example: let’s say I have a beam shaping surface which takes a Gaussian beam and generates a uniform intensity or irradiance disk at some far field plane ... However the field amplitude should be uniform at the target plane right? ( since the intensity or irradiance is the square of the field amplitude)

I've seen lenses with a large amount of spherical aberration capable of something like this, so I'm not totally unfamiliar with this idea. I think it ends up mostly uniform within a certain disk. I don't think there is a way (without vignetting) to create a perfectly uniform disk. However, you can get pretty close.

> Or better said, how could I generate field amplitude distributions from the ray amplitude information? Do you have any suggestion in mind? (Again, I want to do this purely geometrically for the moment, while neglecting diffraction and coherence effects)

If your beam is Gaussian, then it obeys geometrical propagation laws. I wrote a paper about including Gaussian Beamlet Decomposition into Poke for telescope modeling, if you are interested:
https://www.spiedigitallibrary.org/journals/Journal-of-Astronomical-Telescopes-Instruments-and-Systems/volume-9/issue-04/048003/Hybrid-propagation-physics-for-the-design-and-modeling-of-astronomical/10.1117/1.JATIS.9.4.048003.full 

This is less relevant to your use case of propagating a single Gaussian, but contains some of the information on how you would geometrically propagate a Gaussian beam. (Norman Worku and Herbert Gross wrote extensively on this topic as well). I don't have an exact solution in mind, but I suppose you would need to incorporate the geometric change of the Beam waist as a function of free-space propagation distance. I'm not sure I 100% understand where in your system you want to include free-space propagation effects, but if your beam is Gaussian, it's definitely doable geometrically :^)