Jashcraf/poke #119: Citation on Poke use

kalosu, 2025-02-17:
Hi there friends from Poke,

I have used Poke to run some simulations and I would like to include a citation on your work on the manuscript that I am preparing.

How should I cite your project in this case? 

Thanks for the feedback!

Jashcraf, 2025-02-19:
Ah happy to hear Poke was able to help out your simulations :) it's been great seeing you in the issues helping out other users!

For now we ask that people cite the SPIE Proceedings:

https://www.spiedigitallibrary.org/conference-proceedings-of-spie/12664/1266404/Poke-an-open-source-ray-based-physical-optics-platform/10.1117/12.2678001.short

Jashcraf, 2025-02-19:
Marking this as a todo to add to the read the docs