Jashcraf/poke #120: Request for a pure GBD-based PSF simulation example without zosapi

BecIcan, 2025-02-19:
Hi Jashcraf,

I am currently researching end-to-end optimization in optical systems and very interested in incorporating the Gaussian Beamlet Decomposition (GBD) algorithm from the Poke project into my front-end optical simulation workflow.

However, I have encountered a challenge: when using Zemax’s ray tracing algorithm via zosapi, it seems that gradient backpropagation is not supported, which is essential for my optimization work. The existing examples in the project rely on zosapi, making it difficult to integrate with differentiable simulation frameworks.

Would it be possible to provide a self-contained example that utilizes only the GBD algorithm to simulate the PSF without relying on zosapi or any external ray tracing software? A pure Python-based example would be immensely helpful for those of us focusing on gradient-based optimization in optical design.

Thank you for your time and for developing this promising project. I look forward to your insights and any guidance you can provide.

Jashcraf, 2025-02-19:
Hi there and thanks for trying out Poke :D 

Could you tell me a bit more about what you are trying to optimize? Poke can't help you optimize surfaces in Zemax / CODE V with gradient back propagation (I'm not sure that anything can), but it supports a Jax backend for it's own internal polarization ray tracing calculations. The existing GBD module actually operates using finite differences to determine the ray transfer matrix.

If you want to use Poke / GBD with a differentiable framework, you'll need to write a `poke.raytrace` interface with your ray tracer that supports automatic differentiation. See the following for a reference

https://github.com/Jashcraf/poke/blob/fcd2109841908b84dd1a754823af30a683ecb72e/poke/raytrace.py#L10

Effectively, all this does is run a raytrace and then pack the data into numpy arrays in a format that Poke likes, see the [SPIE Proceedings](https://arxiv.org/pdf/2309.04649) for the required attributes:

[image]

If you happen to have a differentiable ray tracer, that would substantially improve Poke's GBD module, which presently uses finite differences to compute the ray transfer matrix. I'd be happy to work with you on a PR if you are interested.

BecIcan, 2025-02-24:
Thank you for your patience in answering my questions. In fact, I want to optimize the optical surface by backpropagating the loss of the PSF. There's something I'm still unsure about—if ray tracing (such as Zemax, which Poke uses) is not differentiable, how can the combination of a ray tracer and GBD achieve automatic differentiation with JAX? Does this mean I need to rewrite a differentiable ray tracer (as you mentioned with poke.raytrace), or is Poke already capable of performing differentiable optimization within the JAX framework ?

BecIcan, 2025-02-26:
Yes, I actually tried optimizing optical surfaces with the current version of Poke, but it threw an error about not finding the propagation object (I suspect this relates to the Zemax API connection step). Thank you for suggesting those project resources!
I’ve examined the core code of Ray Optics and noticed its foundational implementation is based on:
Spencer, G. H., & Murty, M. V. R. K. (1962). General ray-tracing procedure. Journal of the Optical Society of America, 52(6), 672–678.
I’ll study this paper and reimplement a Jax/torch-compatible ray tracer by refactoring the codebase. Appreciate your guidance!

BecIcan, 2025-02-26:
Yes, I actually tried optimizing optical surfaces with the current version of Poke, but it threw an error about not finding the propagation object (I suspect this relates to the Zemax API connection step). Thank you for suggesting those project resources!

I’ve examined the core code of Ray Optics and noticed its foundational implementation is based on:
Spencer, G. H., & Murty, M. V. R. K. (1962). General ray-tracing procedure. Journal of the Optical Society of America, 52(6), 672–678.

I’ll study this paper and reimplement a Jax-compatible ray tracer by refactoring the codebase. Appreciate your guidance!

Jashcraf, 2025-02-26:
Best of luck :D you'll find considerable GBD acceleration with a differentiable ray tracer because you can basically skip the ABCD matrix computation step. 

I recently published a paper on a more memory-efficient implementation of GBD that's implemented in Poke: https://opg.optica.org/oe/fulltext.cfm?uri=oe-32-10-18068&id=549532

With the power of efficient algorithms, automatic differentiation, and GPUs, you should be well on your way to performing some optimization!